sma-bluetooth/sma-bluetooth #138: Doesn't work without mysql (docs say otherwise).

tim-seoss, 2015-05-09:
When leaving the mysqlXXX config items blank, the software still attempts to connect to a mysql server.

stephenbcollier, 2015-05-09:
There will be a new version during next week which will address some of
those issues including 138. I am currently testing it.

koenr, 2015-05-09:
I don't know if it is related, but it also doesn't work with MariaDB on Synology.

tim-seoss, 2015-05-09:
Ah, OK, currently patching this myself, as I could really do with it up and running today.  Any chance you keep the work-in-progress up on github in a different branch?

tim-seoss, 2015-05-09:
Re MariaDB, this is unrelated.

tim-seoss, 2015-05-09:
FWIW, fix here:

https://github.com/tim-seoss/sma-bluetooth

closed 2015-05-09 by tim-seoss

stephenbcollier, 2015-05-09:
I'm using Maria db on El 7 to develop it. What doesn't work?

Cheers
Stephen

koenr, 2015-05-09:
The compiling of the software on the Synology (arm processor).  I'll create
a new and well documented report for it.
Thank you for your interest.
What is El 7?