sma-bluetooth/sma-bluetooth #141: Issues and Questions

artyb5, 2016-02-22:
Hi,
I've been using this software for several years now, but have recently run into issues. Is the project still active? It seems to be flagging? 

I've compiled the present git version, but had to add -I/usr/include/libxml2 to the gcc line for smatool.c. Updated compile instructions would be helpful. Does the programme use the conf file in etc or in the compile directory by default? When I first ran this version of the programme I got a message that I needed to update the database, which I did, apparently successfully. It's now using schema 4. I do get data added to the mysql database, but the dates used when smatool runs are repeated and cludged together eg: 
Auto set dates from 2016-02-22 13:25:02016-02-22 16:20:00 to 2016-02-22 16:20:00

I'm not getting any PVOutput uploads, despite trying variations of --repost and -from and -to dates (these are mis-formatted too sometimes). I've been unable to repost some older data that had been missed, or new data.

Presently I've been running with -d and -v options to try to see what is going wrong and the last lines shown are:
Before todays Almanac
Before Check Schema
SELECT data FROM settings WHERE value='schema' 
SELECT if(sunrise < NOW(),1,0) FROM Almanac WHERE date= DATE_FORMAT( NOW(), "%Y-%m-%d" ) 
SELECT if( dd.datetime > al.sunset,1,0) FROM DayData as dd left join Almanac as al on al.date=DATE(dd.datetime) and al.date=DATE(NOW()) WHERE 1 ORDER BY dd.datetime DESC LIMIT 1
Before close

before update to PVOutput

Then it just hangs.

I'm happy to provide more info to help with debugging. I'm using mainly Debian stable (Jessie) with some backports.

A few years ago I posted a question on the Google Code forum about the patch someone had supplied to try to get it to run (didn't know which version it was patched against, the aim was to be able to reupload a large data-set whilst remaining within the PVOutput limits. I eventually emailed some people involved in the project, and learned of a bereavement, of which I was sorry to hear. The question was never accepted though. It would be a useful thing to be able to do to ensure there are no gaps in a data set.

Hope to hear positive responses.
Thanks,
A

artyb5, 2016-04-02:
Hi,
So, judging by the lack of replies and other coding activity this project doesn't seem to be very active, which is a shame. Maybe people haven't followed the move from google code? Is anyone able to assist, or should I start to look for alternative software?
Thanks,
A

CameronNemo, 2017-12-21:
@artyb5 I don't think this project has a maintainer. It was automatically migrated from google code, meaning the maintainer at google code didn't take any action to migrate it.

rob-hills, 2017-12-23:
Hi,
I think this project has been created by a person who needed it for his own system and it has been modified/updated by other people who used it for their own systems.  I believe that for most of us it is working without any problems.

I had a problem several years ago that necessitated branching the project which I did.  My modified version has been working for my system without any issues for the past 5 years at least.

So, I guess this is a DIY project.  If it doesn't work for you, you roll up your sleeves, or look for something else.  Last time I looked however, there wasn't anything else that worked as well (for me) as sma-bluetooth.

HTH,
Rob Hills
Waikiki, Western Australia

rob-hills, 2017-12-23:
I should mention BTW that one of the reasons I branched from the main project was difficulty debugging issues.  In case it helps you, my branch is here:

[https://github.com/rob-hills/rcphills-sma-bluetooth-dbabstraction](https://github.com/rob-hills/rcphills-sma-bluetooth-dbabstraction)

It includes fairly detailed documentation about setting it up (on Ubuntu, hopefully Debian wouldn't be too different) and clearer error messages for some of the basic configuration traps that can occur.  My knowledge of C is very limited so I may not be able to help much if debugging/recoding is needed, but maybe better error messages will help you resolve your problem?

Cheers,
Rob Hills
Waikiki, Western Australia