sma-bluetooth/sma-bluetooth #143: Smatool Issues

Glutonax, 2017-06-06:
Hi,

Is anyone still using smatool, or is looking for help to use smatool ?

rob-hills, 2017-06-06:
Hi,

Don't need any help with it, it is running happily on my computer as it
has been for the past 4-5 years...

Glutonax, 2017-06-07:
Maybe you can help me. I'm trying to use it with 2 inverters. It work fine with only one but for two i need some modification in database and in the program obviously.

artyb5, 2017-07-19:
Hi,
I've just about given up, my posts below called 'Issues and Questions' led to no response. Are you able to have a look at them? Many thanks.
A

rob-hills, 2017-07-20:
Hi Artyb5,
I see no "posts below" here and none anywhere labelled "Issues and Questions", so I guess you've attached this comment to the wrong thread.  Maybe provide a link to your original post and someone might be able to help.

artyb5, 2017-07-20:
Hi Rob,
Issue #141 is what I was referring to. Best Wishes,
A