sma-bluetooth/sma-bluetooth #144: Can't compile on Debian Stretch

gr0mit1, 2018-06-24:
Hi
I'm trying to build this on a new Debian Stretch 64 bit machine.  I get the following build error:

root@wendolene2:/usr/src/sma-bluetooth# make smatool
gcc -O2 -c smatool.c
In file included from smatool.c:36:0:
/usr/include/libxml2/libxml/parser.h:15:31: fatal error: libxml/xmlversion.h: No such file or directory
 #include <libxml/xmlversion.h>
                               ^
compilation terminated.
Makefile:9: recipe for target 'smatool.o' failed
make: *** [smatool.o] Error 1

I'm sure I'm missing some xml library or other but I can't for the life of me work out what. I've (as far as I know!) got the libxml2-dev installed.

Anyone advise, please?

stephenbcollier, 2018-06-24:
I'll check. I'll run up a vm and check.

gr0mit1, 2018-06-25:
Hi Stephen
Thanks for this.

I'm actually not sure which version I should be using!  There is a
version on github and also aversion 0.18 on google code.

The Google Code version .18 compiles ok, and reads from my inverter but
doesn't seem to update the pvoutput.org details.  Any ideas?

Best Regards

Tim Robinson
TxRx Communications Ltd
+44 1256 810630

gr0mit1, 2018-06-25:
and here is the http conversation with pvoutput.org from a pcap of the
transaction. Looks like it is trying to send older data than pvoutput
wants.  How can we fix this?

GET
/service/r2/addbatchstatus.jsp?data=20180527,05:55,2,24;20180527,06:00,4,24;20180527,06:05,10,72;20180527,06:10,15,72;20180527,06:15,23,96;20180527,06:20,35,144;20180527,06:25,45,120;20180527,06:30,54,120;20180527,06:35,66,144;20180527,06:40,80,144;20180527,06:45,92,144;20180527,06:50,99,96;20180527,06:55,111,144;20180527,07:00,125,168;20180527,07:05,138,168;20180527,07:10,160,240;20180527,07:15,179,240;20180527,07:20,193,168;20180527,07:25,207,168;20180527,07:30,222,192;20180527,07:35,246,264;20180527,07:40,267,264;20180527,07:45,287,240;20180527,07:50,306,240;20180527,07:55,318,144;20180527,08:00,330,120;20180527,08:05,342,144;20180527,08:10,359,216;20180527,08:15,379,240;20180527,08:20,404,312&key=2d3a59f2b5538514e01addd4318f094c754fdcd5&sid=2173
HTTP/1.1
Host: pvoutput.org
Accept: */*

HTTP/1.1 400 Bad Request
Date: Mon, 25 Jun 2018 08:23:47 GMT
Set-Cookie: JSESSIONID=1xdq3s3bbu8tt13gh91dd8i1va;Path=/
Expires: Thu, 01 Jan 1970 00:00:00 GMT
Content-Type: text/plain;charset=ISO-8859-1
Connection: close
Server: Jetty(7.6.17.v20150415)

Bad request 400: Date is older than 14 days [20180527]

Best Regards

Tim Robinson
TxRx Communications Ltd
+44 1256 810630

gr0mit1, 2018-06-25:
Ok - sorted now, needed to delete some of the older data in order for it
to be able to upload.

Still interested in which version I should be using, though!

Best Regards

Tim Robinson
TxRx Communications Ltd
+44 1256 810630

stephenbcollier, 2018-06-25:
github is the correct version the google code has been shutdown I'm 
surprised its still up.

gr0mit1, 2018-06-25:
Hi Stephen

On 25/06/2018 10:42, stephenbcollier wrote:

> github is the correct version the google code has been shutdown I'm
> surprised its still up.
> 

Thanks for clarifying. Can you reproduce the compilation errors I was
seeing with Debian Stretch on the github version?

Best regards
Tim Robinson

ryannathans, 2018-08-17:
Ubuntu 16.04, can't compile

$ make
gcc -O2 -c smatool.c
In file included from smatool.c:36:0:
/usr/include/libxml2/libxml/parser.h:15:31: fatal error: libxml/xmlversion.h: No such file or directory
compilation terminated.
Makefile:9: recipe for target 'smatool.o' failed
make: *** [smatool.o] Error 1

ryannathans, 2018-08-17:
Fixed
```

cd /usr/include
sudo ln -s libxml2/libxml libxml

```

takis, 2021-03-20:
Hi,

I've created a pull request to fix the compilation without symbolic links: #145